selfcustody/krux #431: [Bug] BIP85 "Load child?" translations doesn't fit screen

tadeubas, 2024-07-31:
#### Summary

[image]

[image]

**Device**
- [ ] Amigo
- [x] M5stickV
- [ ] Dock
- [ ] Bit
- [ ] Yahboom
- [x] Cube

**Version**
- [x] Official release (selfcustody/krux): v24.07.0 
- [ ] Beta release (odudex/krux_binaries):  

#### Describe the bug

BIP85 "Load child?" translations doesn't fit some device's screen.

Translations affected:

- fr-FR
- nl-NL
- pl-PL
- ru-RU
- tr-TR
- vi-VN

tadeubas, 2024-07-31:
My suggestion is to simplify this string to "Load?" like we have on other screens

odudex, 2024-08-05:
Solved with #433

PR #433 (2024-08-05): Optmizations
- Consts for special chars
- Code reuse
- Translations fixes for fr-FR (thx @jdlcdl !)
- Fixes #431 
- Simplified github PR and issues templates

### What is the purpose of this pull request? 

- [ ] Bug fix
- [ ] New Feature
- [ ] Documentation update
- [x] Other
commit 94dc4a98 (2024-08-05): "load one" to "load" - fixes #431
closed 2024-08-05 by odudex