selfcustody/krux #444: [Bug] Settings Hardware>Printer>Driver stop working when using TOUCH and PAGE

tadeubas, 2024-08-16:
#### Summary
When changing the `Hardware>Printer>Driver` by TOUCH then pressing PAGE or PAGE_PREV make the config to stop working

**Device**
- [x] Amigo
- [ ] M5stickV
- [ ] Dock
- [ ] Bit
- [x] Yahboom

**Version**
- [x] Official release (selfcustody/krux): 
- [ ] Beta release (odudex/krux_binaries):  

#### Describe the bug

Go to `Settings>Hardware>Printer>Driver`, TOUCH to move next to `cnc/file` then press PAGE pr PAGE_PREV and `None` will be shown, then the config doesn't respond to TOUCH or PAGE/PAGE_PREV anymore. If you click on Go after the config problem, it will be frozen and you will be unable to change it. Only using `Factory Settings` to restore will allow to config this setting again.

jdlcdl, 2024-08-16:
Do the same in language, and you'll get locale=None as well.  Thank you for figuring out how to provoke that!

tadeubas, 2024-08-16:
Odudex suggested this https://github.com/selfcustody/krux/blob/develop/src/krux/themes.py#L130

```
if current_theme not in (list(ThemeSettings.THEME_NAMES.values())):
            # In case old version theme in use was deleted
            current_theme = ThemeSettings.DARK_THEME_NAME
            Settings().appearance.theme = current_theme
```

odudex, 2024-08-16:
> Odudex suggested this https://github.com/selfcustody/krux/blob/develop/src/krux/themes.py#L130
>
> ```
> if current_theme not in (list(ThemeSettings.THEME_NAMES.values())):
>             # In case old [...]

This was before you found out how to reproduce the error.
I'm pushing a fix now.

odudex, 2024-08-16:
Fixed with af4b62157ff94a066071101ca41e1d357d0902b6

closed 2024-08-16 by odudex