selfcustody/krux #467: [Bug] Problems using camera to scan QR from Coldcard Q screen

nervetrip, 2024-10-11:
No matter what I do, I can't get Krux running on Yahboom to scan any QR codes displayed on a Coldcard Q.

Here's a throwaway seed phrase on the Q with screen brightness turned to the minimum (25%) and a desk lamp illuminating the area:
[image]

Krux refuses to focus properly when scanning the Q screen. I've tried so many times with different distances, angles, screen brightness, ambient light levels, etc. These pictures illustrate what I'm seeing on the Yahboom screen. Note that my fingers are fully in focus, but Krux can't seem to focus on the QR code. I took the pics when it was as good as I could get it, it certainly was often even worse.
[image]
[image]

However, Krux scans QR codes from my computer screen without any issues. The QR code is fully in focus on the Krux screen:
[image]

Lastly, I can take a picture of the QR code on the Coldcard screen, display it on my computer, and then Krux successfully scans it. However, it scans all the seed words in capital letters, and indicates that the 12th word is invalid.
[image]

I'm not sure if this qualifies as a bug, but I'm hoping firmware/camera tweaks might improve QR code scanning with this scenario. Any other thoughts on how to improve scans?

jdlcdl, 2024-10-11:
> I'm thinking that loading this style of QR code content is probably a feature enhancement for Krux.
 
 I agree, it looks like the qrcode represents first 4 chars of each word in uppercase (while krux can scan complete words in lowercase, Im not aware that it knows to downshift case and to extrapolate the full word... yet.
 
 Does the yahboom have a camera lens wheel where you can fiddle with focus?  If it does, that might be the trick for reading the q screen. ???
 
 Nice write-up!

jdlcdl, 2024-10-11:
>  it would help a lot if Q could increase the QR frame to the recommended 2 QR blocks (currently 1)

Ahhh, Now I see it.  The full-white quiet-zone margin around the qrcode contents.

odudex, 2024-10-11:
[Krux 24.10.beta4](https://github.com/odudex/krux_binaries) can now read Q QR backup. At least from computer screen :grinning:
Unfortunately the devices cameras are low quality, so to read from Q you would need bring the focus very close, or use magnifying lenses, so your focus keeps well adjusted for computer screen.
On the Q side, a wider white frame or an upscale would help.

nervetrip, 2024-10-11:
Thanks for the help!

> By rotating the camera lenses (this can be tricky at first)

There's a ring with tiny notches around the lens on the Yahboom, but I can't tell if I'm spinning it. Is there a trick to twisting it? Are needlenose pliers needed, or is it easy to spin and I just can't tell its spinning?

> Anti-glare, activated by pressing the bottom button while scanning with Yahboom, could also help in these cases. 

I just tried this and it maybe made a slight difference in the appearance, but didn't solve the problem. I also tried adding a magnifying glass, but couldn't get that working either. I was able to get slightly more "square pixels", but the magnifying glass was warping the QR which may have been an issue.

> However, Krux can't read this standard (chopped uppercase words). 

Thanks for fixing this so quickly! I just flashed the new firmware and confirmed it works - also from a computer screen, not from the device.

> But it would help a lot if Q could increase the QR frame to the recommended 2 QR blocks (currently 1) or, even better, use the full screen to show the QR.

I can't seem to locate where to report Coldcard bugs and feature requests. Does anywhere know?

tadeubas, 2024-10-12:
I've submitted an e-mail to Coldcard support with a link to this issue, asking for a QR increase in size and wider white on external borders.

odudex, 2024-10-12:
> There's a ring with tiny notches around the lens on the Yahboom, but I can't tell if I'm spinning it. Is there a trick to twisting it? Are needlenose pliers needed, or is it easy to spin and I just can't tell its spinning?

It's a risky operation. The lenses are threaded, so by rotating them, you move them back and forth. They put a small drop of glue on this thread, so the best thing to do is remove the Yahboom's back lead, find and remove the glue with a toothpick or an exacto knife. Then delicately use needle-nose pliers to rotate the lenses. You may need a second pair of pliers to hold the camera's base. Be careful; everything is delicate, from the cable to the lenses.

nervetrip, 2024-10-16:
> It's a risky operation. The lenses are threaded, so by rotating them, you move them back and forth. They put a small drop of glue on this thread, so the best thing to do is remove the Yahboom's back [...]

This is very helpful. I haven't tried it yet, but maybe I'll work up the guts at some point once I'm frustrated enough with CC QR codes not scanning. I'm not holding my breath that Coinkite will resolve the issue on their end. 

Do QR codes on the Coldcard Q scan without issues on any of the other devices Krux is compatible with? Maybe I should try another device. I'd like to scan from the Q, as it would improve my workflow.

odudex, 2024-10-16:
https://github.com/user-attachments/assets/086e5854-3430-4b42-b73b-9ce3b2ec903d

If you bring focal distance closer, it will easily scan.

nervetrip, 2024-10-18:
@odudex Perfect, your video looks like exactly what I'm hoping for!

I took my Yahboom apart and attempted to remove the glue with an exacto knife. I think I got all of it, plus some of the camera casing is chewed up a bit. There could be some glue left in-between the base and the ring, its hard to tell.  

But I still couldn't get it to twist. It almost seemed like it may be one piece and won't twist, but it's really hard to tell. I didn't put a ton of pressure on twisting because I didn't want to break it. 

So I put it back together and its still the same as it was. 😢  I may try again later.

odudex, 2024-10-18:
Great!
Once you release the thread, it becomes easy to move; usually, you can do it with your fingers.
Focus is a recurrent issue we have. Autofocus would be a great upgrade to Krux, and there are autofocus compatible cameras, I may try one soon.

closed 2025-02-26 by nervetrip