selfcustody/krux #468: [Enhancement] Add additional BIP-85 applications, especially Base64 Passwords

nervetrip, 2024-10-12:
It would be really nice if more BIP-85 applications were supported in Krux. I would especially appreciate the BIP-85 "PWD BASE64" application for password generation. Here's the relevant section of the BIP-85 spec:
https://github.com/bitcoin/bips/blob/master/bip-0085.mediawiki#pwd-base64

But I still want easy access to the 

For reference, Coldcard can be used to generate 21 character passwords instead of seed words for each BIP-85 index: https://coldcard.com/docs/bip85-passwords/

Coldcard currently offers the following BIP-85 applications (Krux offers only the first two):
- 12 words
- 24 words
- WIF
- XPRV
- 32-byte hex
- 64-byte hex
- password

nervetrip, 2024-10-12:
It looks like embit doesn't include a feature to derive a password:
https://github.com/diybitcoinhardware/embit/blob/master/src/embit/bip85.py

But Coldcard's implementation seems fairly simple: 
https://github.com/Coldcard/firmware/blob/master/shared/drv_entro.py#L101

odudex, 2024-10-14:
This can indeed be very helpful!
Coldcard can simulate keystrokes through USB, but this would not be possible for most of our devices. Would it suffice to show the passwords on the screen?
As QR codes, it could be dangerous; people would likely need to pass it through third-party apps before using it to log in.
I would also add a step for choosing length, defaulting to 21 to match CC.

odudex, 2024-10-14:
Available to test on 24.10.beta5!

nervetrip, 2024-10-15:
Brilliant, it appears to work perfectly! This will be so helpful, appreciate the effort. 

And variable password lengths makes it even better than Coldcard's implementation. Always using passwords of 21 characters could provide hints to an attacker that they are potentially BIP85 passwords.

I agree that QR codes for passwords might be risky for some use cases, but it seems like a mistake to not include the QR display feature. Let users decide if they want to use it or not based on their situation. Being able to scan the QR code would absolutely simplify my completely offline air gapped use case. 

If it isn't too much effort to display passwords as QR codes, I personally would prefer having it as an option.

odudex, 2024-10-15:
>Let users decide

You know how to convince me 😁

odudex, 2024-10-15:
24.10.beta6: Allow export BIP85 passwords to SD and as QR code

nervetrip, 2024-10-16:
> You know how to convince me 😁

Good to know! 😜 Thanks for adding this, will make a world of difference. Any thoughts on when it might make it into an official release?

nervetrip, 2024-10-16:
I just realized there may be one more thing necessary to complete this feature. Setting a passphrase for a wallet only has the options to `Type BIP39 Passphrase` and `Scan BIP39 Passphrase`. Would be ideal to enter and scan BIP85 passwords (21 characters or any valid length) as well as BIP39 seeds.

jdlcdl, 2024-10-16:
Would be ideal to enter and scan BIP85 passwords (21 characters or any valid length) as well as BIP39 seeds.

I suspect it will magically work as intended.  While the qrcode will originate from a bip85 password, it will simply decode into text which is no different than if you'd created the password with tools/create-a-qrcode.  You'll be able to use it like an encrypted-mnemonic key, or a bip39 passphrase, or anywhere else that is expecting to decode a qrcode into text.

jdlcdl, 2024-10-17:
> The labels are deceptive since they say BIP39...

Please elaborate on this.  When I try to create a bip85 base64 password, I am proposed to save the file to sdcard as "BIP85_password.txt" which seems correct to me... it is a default hint of where the password came from.  When viewing the qrcode, it's just "Base64 Password" below the qrcode image...which also seems correct.  When using Tools / Create QR Code, I get a label of "Custom QR Code" under the qrcode image, and can save to sdcard as Custom_QR_.pbm

nervetrip, 2024-10-18:
> the labels "Scan BIP39 Passphrase" or "Type BIP39 Passphrase" are an important distinction...it is going to be applied as a BIP39 passphrase...

This makes total sense now. I was reading `BIP39` but was thinking `BIP85`. Keeping these labels as is makes the most sense. Please forgive my confusion.

nervetrip, 2024-10-24:
@odudex Did something change between beta5 and beta6? The passwords generated by Krux are no longer the same as the passwords generated by my Coldcard Q.

I'm 99% sure that I verified they were the same when you first added the feature, right before I posted my "Brilliant!" comment. I can't imagine that I wouldn't have tested it. But I may not have double-checked again after installing the beta6 release.

### Steps to Reproduce

I'm using this throwaway QR code as the starting BIP39 for my tests:
[image]

On Krux:

- Scan QR code to load mnemonic and load wallet with fingerprint **DC458784**
- Go to Wallet -> BIP85 -> Yes -> Base64 Password
- Enter `1` for the Index
- Continue with `21` for the length
- Result is `y4lmlTja+nYpxA7rkYcMq`

On Coldcard:

- Press QR button and scan QR code to load mnemonic as temporary wallet with fingerprint **DC458784**
- Go to Advanced/Tools -> Derive Seeds (BIP85) -> Enter -> Enter -> Passwords
- Enter `1` for the Index
- Result is `1pgbLa65cqqdj6iCw+i84`

odudex, 2024-10-24:
I'm so sorry, I added a bug during a refactor, omitted the length from the derivation path. I was sloppy not using external references for the tests I later implemented.
Bug should be fixed in 24.10.beta9
I will warn users on Telegram and Twitter for whoever eventually used this beta for a real password to note it down then replace it.
I will also improve tests with eternal references. Thank you again for the test and detailed report!

nervetrip, 2024-10-24:
Latest build works like a charm! Thanks for the fix.

jdlcdl, 2024-10-24:
> should have used external references for the tests

I tested the first version by hand.  The "external reference" testcases for bip85 were actually wrong at the time... for about a 1 week span (the password was correct but the entropy was wrong and according to the bip, both are supposed to pass).

I should have caught your refactor, but I didn't.  Not sure I would have caught it before release either (more to review is more to miss).  I'm sure glad that @nervetrip caught this early!!!  🙏

tadeubas, 2025-02-26:
Hey @nervetrip I think this feature was release on [v24.11.0 ](https://github.com/selfcustody/krux/releases/tag/v24.11.0).

Plz close this issue it is resolved, or update with what is still missing. Thx!

closed 2025-02-26 by nervetrip