selfcustody/krux #473: [Enhancement] Make beta builds installable airgapped via SD card

nervetrip, 2024-10-24:
@odudex I've been installing your beta builds via USB with ktool. This works fine but I'd like to eventually be fully airgapped.

Is it possible to include a `firmware.bin.sig` file whenever a new build is created? I tried installing with just `firmware.bin` on the SD Card, but it failed. Or am I doing something wrong? I'm following the instructions here:
https://selfcustody.github.io/krux/getting-started/features/sd-card-update/

odudex, 2024-10-24:
Many users have two or more devices and use one of them for more serious use, flashing only stable releases and keeping it always air-gapped. The other device is used to help us test new features, satisfy curiosity, and report bugs and feedback. As you have been doing in an exemplary way (thank you 🙏).
The official signature helps keep things separated. 
I could do better structured beta releases, but I would not use the selfcustody signature for them. Maybe I could create another one, which requires a less secure and more practical protocol, and when firmware detects it is a beta signature it warns the user. What do you think?

nervetrip, 2024-10-24:
That's a great point! It does make sense to keep the official signature of stable releases isolated from unstable releases. I hadn't really thought through the implications when making this request.
 
I should probably get another device. So far I've been using it only for experimenting and testing, so installing beta versions hasn't been a concern. My first instinct is to say not everyone can afford multiple devices, but I do realize that most contributors and people willing to test beta releases can.

Installing beta releases in an air-gapped way would be nice, but it isn't critical and shouldn't be high priority. If you think there's value to be had, then an alternate signature for beta releases seems like a good solution. And a firmware warning would add a nice touch.

tadeubas, 2025-04-30:
@nevertrip thanks for raising this! If you have any suggestions to improve our documentation on this, please let me know!

The SD card, air-gapped update flow is meant as an extra security layer **only** for official, stable releases - those come signed with `firmware.bin.sig`. Beta builds are designed purely for quick feature testing under a more flexible trust model, so we don’t apply the official signature to them as we can push multiple builds in a single week for just a handful of testers. Also all signing is done air-gapped, one file at a time, using a manual, hardware-backed (Krux) process - so it’s labor-intensive. Automating with a separate key to beta is possible, but will require some code changes and we are about to enforce version checks and prevent installing older firmware with PR #583 - so IMHO beta should not be applied to SD card update.

What do you think? Can we close the issue?

nervetrip, 2025-05-02:
Your argument makes complete sense, and I certainly am not wanting to complicate or slow down your development process. If this isn't something you find valuable for krux, feel free to close the issue. However, let me first provide some background for consideration. 

I'm currently on an extended 6-12 month trip around the world and have zero "extra" room in my bag to carry multiple devices. I purchased the Yahboom because it was small and compact, yet still fully functional with touchscreen and isn't immediately recognizable as a "crypto wallet" if a border agent conducts a search. This is why I'm especially interested in #203 for plausible deniability regarding the purpose of the device.

I've been flashing it via USB since that's the only way to test beta releases. But at some point I plan to use it to make real transactions from my cold wallet. Once I have loaded my cold seed phrase onto the device, I don't want to ever connect it via USB again. So swapping between beta and release versions would need to be done in an air gapped way.

Now that I think about this deeper, I realize its foolish of me to even consider using a single device for both beta testing and development as well as cold wallet production use. I probably should have two devices. Since I have an android phone, maybe I could use it to test beta releases instead.