selfcustody/krux #477: [Bug] Encrypted QR codes with different i_vector in AES-CBC mode are identical.

earthdiver, 2024-11-19:
#### Describe the bug
When creating an encrypted QR code in AES-CBC mode, we are obtaining an additional entropy from the camera, but the generated QR codes are the same for the same combination of key, mnemonic_id, and mnemonic. I believe that the first block in AES-CBC mode should not be i_vector.

**Device(s) affected**
presumably all devices

**Version affected**
presumably all versions

odudex, 2024-11-19:
Thank you for your important finding! It helped me identify an error in the CBC encryption implementation, which unintentionally "deactivated" i_vector. I'm ashamed of not performing similar test before.
We've made a dedicated release for the fix.
Thank you again for the catch!

tadeubas, 2024-11-19:
Release with the fix:
https://github.com/selfcustody/krux/releases/tag/v24.11.1

odudex, 2024-12-11:
I'll consider this issue solved. Please report otherwise.
Once again, thank you.

closed 2024-12-11 by odudex