selfcustody/krux #486: [Bug] Multisig Bluewallet Wallet Descriptor not recognised

TimYorke, 2024-12-04:
Version 24.11.1 running on yahboom HW

I see the error "Invalid wallet: ValueError(invalid wallet format',)" when I try to import the wallet descriptor from BlueWallet (Android). I see the progress bar increase as it recognises the QR codes, but then just displays a black screen with this error message. 

I can be reproduce it using the QR code animation displayed in "Export Coordination Setup" in BlueWallet, and via "Tools->Descriptor Addresses->Load from camera" in Krux

It's a 2/2 native segwit "Multisig Vault" and derivation path "m/48'/0'/0'/2'"

I'm happy to try and debug locally if someone can give me some pointers.

jdlcdl, 2024-12-10:
Would you be able to confirm if you see the cosigner keys as "Zpub" instead of "xpub" in your BlueWallet backup file?

odudex, 2024-12-11:
Your report helped us identify incompatibilities with BlueWallet multisig descriptors and PSBT signing in Krux. We’ve released a fix in [v25.01.beta3](https://github.com/odudex/krux_binaries), which should allow you to successfully load your descriptor and sign PSBTs.

A stable release containing this fix is planned for next month. After not been testing with Blue recently, we will now do more tests and continue to improve compatibility. In the meantime, if you prefer to stick with the current stable release, you can try recreating your wallet in BlueWallet by exporting only the xpubs. This approach should help you avoid the issue until the stable fix is available.

TimYorke, 2024-12-11:
Thank you. The xpub-only workaround did the trick

odudex, 2025-03-20:
Latest release should have it fixed, can you confirm?

closed 2025-04-09 by odudex