selfcustody/krux #504: [Bug] the summary of your report

rough-street, 2025-02-20:
I've noticed what looks like an important bug in the signing screen amounts.

If I simulate signing a transaction from a compromised device whereby the attacker has modified the change address such that it goes to a different wallet, I notice that on the initial screen after reading the transaction, with "Inputs", "Spend" and "Fee", then the Spend amount is correctly wrong - it shows the full input amount (minus fee), but on the next screen it is shown as different figure - in fact the one I'd expect if the transaction had a legitimate change address owned by the wallet. 

In other words, the second "1. Spend:" screen is not detecting that the change address does not belong to the wallet and gives the impression the tampered-with transaction is fine. 

**Device(s) affected**
- [ ] Amigo
- [ ] M5stickV
- [ ] Cube
- [ ] Dock
- [ ] Bit
- [*] Yahboom
- [ ] WonderMV

**Version affected**
- [*] Official release (selfcustody/krux): v24.11.1
- [ ] Beta release (odudex/krux_binaries): vXX.YY.Z-betaWW

odudex, 2025-02-20:
@rough-street I recently changed change detection to not trust the PSBT data, can you confirm in [latest beta](https://github.com/odudex/krux_binaries) the issue is solved?

rough-street, 2025-02-20:
After continuing the test, I now realise that there are now more than one "spend" screen after the main PBST summary screen, and each screen details a spend for each address involved. In my simulated attack, I now notice that I actually see two spend screens - one for the desired destination, and then a second for the modified, incorrect change address. So there's no bug after all - just my mistake - Krux is working correctly.

closed 2025-02-20 by rough-street

odudex, 2025-02-20:
Great! But we did change the change detection for beta and for next release to be more restrictive. Attack tests like this with are really welcome! We could also implement them on auto-tests.

rough-street, 2025-02-20:
Sounds good - I'll be testing it as soon as it comes out. Loving your work BTW