selfcustody/krux #550: [Bug] Generating qr.nc fail : Error: IndexError('bytes index out of range',)

spatcho, 2025-03-28:
#### Describe the bug
Printer driver is set to cnc.

It doesn't matter if you load the wallet from a new mnemonic or from a qr scan.

After wallet is loaded, I do : Backup Mnemonic > Encrypted > Encrypted QR Code > Type key and proceed without custom id

When presented with QR code on device screen, I do : Print as QR > yes to "cnc/file" > and I quickly get a screen showing in red "Error: IndexError('bytes index out of range',)"

The sd card is recognized, I could save qr as images, and I see an empty "qr.nc" file with updated timestamp but 0 bytes size.

Issue happen on my wonder my and on the simulator for wonder my.

I am now trying to build a firmware with MICROPY_ENABLE_COMPILER to see more infos and I will try the simulator with other types of devices. I have a cnc on grbl, that would be another step.

**Device(s) affected**
- [ ] Amigo
- [ ] M5stickV
- [ ] Cube
- [ ] Dock
- [ ] Bit
- [ ] Yahboom
- [X] WonderMV

**Version affected**
- [X] Official release (selfcustody/krux): v25.03.0
- [ ] Beta release (odudex/krux_binaries): vXX.YY.Z-betaWW

odudex, 2025-03-28:
Export to CNC feature is "abandoned", as non of devs have one to test, and, until now, we didn't know any user making use of it. I was even suggesting to remove it.
But since we have you now, and it seems you're motivated to make it work, we can try to help!
I'll try to track and solve the error on my side too

tadeubas, 2025-03-28:
I don't think you need to enable compiler, just comment [these lines](https://github.com/selfcustody/krux/blob/main/src/krux/pages/__init__.py#L638) of the try/except and build the firmware using the script `/.krux build maixpy_yahboom` (or other device name) and flash the firmware onto device using instructions [here](https://selfcustody.github.io/krux/getting-started/installing/from-source/). Then follow [these instructions](https://selfcustody.github.io/krux/troubleshooting/#device-didnt-reboot-and-screen-is-blank) to use MaixPy IDE and see the exception on terminal.

spatcho, 2025-03-28:
I am on silicon Mac but I could compile and build a new firmware using the linux how to : krux build maixpy_wonder_mv . To flash I didn't find out how to pass the -p and -B params to krux flash so I used odudex ktool-mac, replacing the contents of maixpy_wonder_mv with my binaries : ktool-mac -p /dev/cu.usbserial-xxxxxx -B dan -b 1500000 maixpy_wonder_mv/kboot.kfpkg . Seems it went through, I see my updated version in the about screen on device.

In this [section](https://github.com/selfcustody/krux/blob/main/src/krux/pages/__init__.py#L638), I removed line "self.ctx.display.clear()", but it doesn't help having more details on the issue.

About MaixPy IDE, I tried, seems it connect as the device is rebooting, but it never complete and the device stay frozen. As I see no "wonder mv" in the supported devices list, could I do something ?

tadeubas, 2025-03-29:
Change this to see the error in the terminal -> remove the `try / except` block.

From this:
```
try:
    self.ctx.display.clear()
    status = item[1]()
    if status != MENU_CONTINUE:
        return status
except Exception as e:
    self.ctx.display.to_portrait()
    self.ctx.display.clear()
    self.ctx.display.draw_centered_text(
        t("Error:") + "\n%s" % repr(e), theme.error_color
    )
    self.ctx.input.wait_for_button()
```

To this:
```
self.ctx.display.clear()
    status = item[1]()
    if status != MENU_CONTINUE:
        return status
```

Then to connect to the device and see the output in terminal copy the below file (`kboot.py.txt` rename to remove the .txt at the end) to the project root dir and execute (maybe you will need to install dependencies, use `poetry install --all-extras` to install everything including dependencies needed to run our simulator):
`poetry python ktool.py -t`
if you don't see nothing changing in the terminal try to reset the device while connected or to use the cmd with this param:
`poetry python ktool.py -t -B dan`

[ktool.py.txt](https://github.com/user-attachments/files/19515893/ktool.py.txt)

If you want to test faster, use our simulator with sd card enabled (a folder `sd` will be created on the project's `simulator` folder):
`poetry run poe simulator --sd`

I've tested on the simulator and can confirm that this error appears:
[image]

The stacktrace below appears on terminal (with the change mentioned above) :
```
Exception in thread Thread-1 (run_krux):
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/krux/simulator/simulator.py", line 130, in run_krux
    exec(boot_file.read())
  File "<string>", line 159, in <module>
  File "<string>", line 133, in home
  File "/krux/src/krux/pages/__init__.py", line 512, in run
    _, status = self.menu.run_loop(start_from_index)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/__init__.py", line 686, in run_loop
    status = self._clicked_item(selected_item_index)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/__init__.py", line 718, in _clicked_item
    status = item[1]()
             ^^^^^^^^^
  File "/krux/src/krux/pages/home_pages/home.py", line 71, in backup_mnemonic
    return mnemonics_viewer.mnemonic()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/home_pages/mnemonic_backup.py", line 46, in mnemonic
    submenu.run_loop()
  File "/krux/src/krux/pages/__init__.py", line 686, in run_loop
    status = self._clicked_item(selected_item_index)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/__init__.py", line 718, in _clicked_item
    status = item[1]()
             ^^^^^^^^^
  File "/krux/src/krux/pages/home_pages/mnemonic_backup.py", line 87, in encrypt_mnemonic_menu
    return encrypt_mnemonic_menu.encrypt_menu()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/encryption_ui.py", line 176, in encrypt_menu
    _, _ = submenu.run_loop()
           ^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/__init__.py", line 686, in run_loop
    status = self._clicked_item(selected_item_index)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/__init__.py", line 718, in _clicked_item
    status = item[1]()
             ^^^^^^^^^
  File "/krux/src/krux/pages/encryption_ui.py", line 280, in encrypted_qr_code
    seed_qr_view.display_qr(allow_export=True)
  File "/krux/src/krux/pages/qr_view.py", line 505, in display_qr
    _, status = submenu.run_loop()
                ^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/__init__.py", line 686, in run_loop
    status = self._clicked_item(selected_item_index)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/pages/__init__.py", line 718, in _clicked_item
    status = item[1]()
             ^^^^^^^^^
  File "/krux/src/krux/pages/qr_view.py", line 435, in print_qr
    utils.print_standard_qr(self.code, title=title, is_qr=True)
  File "/krux/src/krux/pages/utils.py", line 50, in print_standard_qr
    print_page.print_qr(data, qr_format, title, width, is_qr)
  File "/krux/src/krux/pages/print_page.py", line 62, in print_qr
    self._send_qr_to_printer(data)
  File "/krux/src/krux/pages/print_page.py", line 48, in _send_qr_to_printer
    self.printer.print_qr_code(qr_code)
  File "/krux/simulator/kruxsim/mocks/machine.py", line 40, in new_print_qr_code
    return old_print_qr_code(qr_code)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/krux/src/krux/printers/cnc.py", line 255, in print_qr_code
    super().print_qr_code(qr_code)
  File "/krux/src/krux/printers/cnc.py", line 72, in print_qr_code
    while qr_code[size] != "\n":
          ~~~~~~~^^^^^^
IndexError: bytearray index out of range
```

## Conclusion
For me it appears that CNC driver was built to deal with TEXT QRCodes only, but the Encrypted one is in binary form.

spatcho, 2025-03-29:
Thanks for help, with your code modification suggestion and that command "poetry run python ktool.py -t -p /dev/cu.usbserial-xxxx -B dan" I could see output clearly. 

For some reason in my initial testing I saw the cnc option only for encrypted, but now I see the option for any kind of qr.

The error is also with plain text qr, seed qr, compact qr.

spatcho, 2025-03-29:
The simulator, can it keep the printer driver settings (default to cnc, etc) and load a qr from an existing source ? Right now to use it I have to set the driver every time and fastest I found is to create a mnemonic with 12 words.

spatcho, 2025-03-29:
I placed some debug infos in that [block](https://github.com/selfcustody/krux/blob/b4de1206f96bd1dc150f307c30817c7710cf6dce/src/krux/printers/cnc.py#L69) and checked for end of qr_code array loop by size instead of '\n' char, to go further without exception.

Did the qr_code array structure changed since the last cnc tested version, around mid June 2023 as of this [issue](https://github.com/selfcustody/krux/issues/171) ? No matter the type of qr (plain or compact tested), the array is filled with numbers. I am no python expert but the code look for a '\n' whithout (I think) converting to char, and to decide whether to cut or not, it look for 0 or 1, which would I guess mean it's a hole in the qr, but there is very few 0 or 1 on the qr_code array.

Can you explain me how the qr_code array work ?

odudex, 2025-03-29:
Sorry I didn't look at this issue yet, but I see you are digging deep.
Yes, the QR structure was optimized, from strings of zeros, ones and `\n`, to binary format, where each QR "pixel", or block, is represented by 1 bit. This is likely the cause of the issue.

tadeubas, 2025-03-30:
Yes, if you run the simulator with `--sd`. This will create an `sd` folder inside `simulator` and enable access to files there. In settings you can change the **Persist -> Location** to **SD card** and this will save any changes you made to defaults.

To test this faster at "Home" screen, I like to change `boot.py` and "jump" `Login` by commenting its line [here](https://github.com/selfcustody/krux/blob/main/src/boot.py#L160), then create a new Wallet in `ctx` (context object):
```
# login(ctx)
# gc.collect()
from krux.wallet import Wallet
from krux.key import Key
ctx.wallet = Wallet(Key("action action action action action action action action action action action action", False))
home(ctx)
```

As you have already identified, you can use an older version in order to test CNC for now. I think we can convert the QR code back to the non "optimized structure" before sending it to the CNC printer. This would solve the issue.

spatcho, 2025-03-30:
SD card and new wallet in simulator, great tip, thanks 👍 

Does it make sense if I inspire from the [thermal printer code](https://github.com/selfcustody/krux/blob/d66f21e1ef6352a4a9bc639fd13b761add6b97db/src/krux/printers/thermal.py#L118) to understand the optimized qr_code array and adapt the cnc code ? I don't have a thermal printer, I don't know if that part is up to date.

odudex, 2025-03-30:
> Does it make sense if I inspire from the [thermal printer code](https://github.com/selfcustody/krux/blob/d66f21e1ef6352a4a9bc639fd13b761add6b97db/src/krux/printers/thermal.py#L118) to understand the [...]

It does, shouldn't be complicated. Thermal printer is working.
Do you have grbl gcode simulator/visualizer to inspect the results?

spatcho, 2025-03-30:
I have an openbuilds cnc that I want to test with, and I can inspect first with open builds control software.

spatcho, 2025-03-30:
Ok, did some progress [here](https://youtu.be/7mH9H314tz0) , should reproduce the "action action ..." wallet in compact format, as of below, it miss some data, I will see 

[image]

tadeubas, 2025-03-30:
Wow, didn't know that this exists https://software.openbuilds.com/ very nice!

odudex, 2025-03-30:
> Ok, did some progress [here](https://youtu.be/7mH9H314tz0) and here

You're close!

spatcho, 2025-03-31:
> > Ok, did some progress [here](https://youtu.be/7mH9H314tz0) and here
> 
> You're close!

Indeed . Now the dots are correctly read.

[image]

```
    def print_qr_code(self, qr_code):
        """Prints a QR code, scaling it up as large as possible"""
        from ..qr import get_size

        size = 0

        size = get_size(qr_code)
        print("size:",size)

        print("self.part_size",self.part_size," / self.border_padding:",self.border_padding," / size:",size)

        cell_size = (self.part_size - (self.border_padding * 2)) / size

        print("cell_size:",cell_size)

        # Modal settings
        self.on_gcode("G17")  # x/y plane
        self.on_gcode("G20" if self.unit == "in" else "G21")  # units
        self.on_gcode("G40")  # cancel diameter compensation
        self.on_gcode("G49")  # cancel length offset
        self.on_gcode("G54")  # coord system 1
        self.on_gcode("G90")  # non-incremental motion
        self.on_gcode("G94")  # feed/minute mode

        print("self.cut_depth:",self.cut_depth," / self.pass_depth:",self.pass_depth)

        #num_passes = math.ceil(self.cut_depth / self.pass_depth)
        num_passes = 1
        for p in range(num_passes):
            print("pass #",p)
            for row in range(size):
                print("row #",row)
                byte = 0
                line_bytes = bytearray()
                for col in range(size):
                    print("col #",col)
                    bit_index = row * size + col
                    print("bit_index #",bit_index)
                    bit = qr_code[bit_index >> 3] & (1 << (bit_index % 8))
                    cut = 0
                    if bit > 0:
                        cut = 1
                    print("bit:",bit," / cut:",cut)
                    if cut:
                        # Flip the y coord
                        y = row
                        x = col
                        x_index = x
                        plunge_depth = min((p + 1) * self.pass_depth, self.cut_depth)
                        self.cut_cell(x_index, size - 1 - y, cell_size, plunge_depth)

```

spatcho, 2025-03-31:
I added an "SVG" option to the "Save QR Image to SD card", this output an .svg, it use lib https://github.com/orsinium-labs/svg.py . That was useful for me to play with 3d printing, and real quick to do now that I got the qr_code format understood. Would it be interesting to add to the trunk ?

Here the encrypted qr as svg  [image]

Am I right that it is only possible to export encrypted qr images and not compact, plain, etc ?

spatcho, 2025-03-31:
I think I am ready to run the nc file with my cnc, it will take me some time, the machine need some additional work first.

odudex, 2025-03-31:
> Am I right that it is only possible to export encrypted qr images and not compact, plain, etc ?

Yes, for now we don't allow saving unencrypted secrets, even in image formats.

I'm impressed on how fast you got familiarized with the code, fixed bugs and implemented new features!
We'll want a Pull Requests!

Also, please consider joining our [Telegram group](https://t.me/SC_Krux). People will love to follow your hacking progress with 3D printer and CNC!

tadeubas, 2025-03-31:
Nice addition! But I think this SVG lib will not work on device, need to test... did you find a micropython implementation alternative for SVG?

spatcho, 2025-03-31:
> Nice addition! But I think this SVG lib will not work on device, need to test... did you find a micropython implementation alternative for SVG?

Ah! Indeed you are right, it fail to fin the svg lib ... the output svg is very simple, I will see to code it.

spatcho, 2025-04-01:
I adapted the svg output to micropython. No additional lib. I don't like to write xml without a support lib but didn't find one for micropython, except a parser which I don't need. So manual write of '<svg xmlns...' etc but I guess the structure is simple enough to be acceptable.

I tested again cnc and svg to sd card on the device, not only on the simulator.

I would like to make the pull request, would you mind to instruct me a bit ?

tadeubas, 2025-04-01:
Wow, nice to know that! To create a PR:

1.  Fork the project from here (selfcustody).
2.  Clone the project into your machine
3.  On the folder create a new branch using `git checkout -b qr_cnc`
4.  Make changes to the files you needed
5. Commit `git commit -m` and type your message about what is in the commit
6. Push the changes to your branch `git push` (for the first time, git will say the branch doesn't exist on your fork yet, only on your machine, so follow the cmd instructions it gives to you to it will be something like `git push --set-upstream origin`) 
7. Your branch `qr_cnc` and changes will appear on Github on your fork (your projects).
8. Navigate to your fork (your project) on github and a button will appear at top suggesting to create a PR on selfcustody based on the branch you recently pushed (created on github), just click on this button and follow instructions selecting the selfcustody `develop` branch (alternatively you can enter in selfcustody/krux click on Pull request at top and then new and select the selfcustody `develop` branch and you `qr_cnc` branch)

You can use VSCode to help with git related commands by using its interface if you prefer

Then we will see your changes on the PR here and comment about it to fix something in order to integrate the code.

spatcho, 2025-04-03:
Could you help me with the rx tx pins on the wondermv ? I connected the hardware correctly but I don't know if the 27 / 28 pin numbers I see on the app are correct and I didn't find any specific specs on wondermv pins setup

spatcho, 2025-04-03:
Thanks a lot. Where did you find those specs ? I ordered a m5stickv and a maix cube, so I can test other devices.

tadeubas, 2025-04-03:
On thermal printer we always recommend to use the TX pin and GND only because Krux doesn't need to receive nothing from the printer, and connecting RX sometimes could cause issues.

odudex, 2025-04-03:
> Where did you find those specs ?

After intensive investigation :)
https://drive.google.com/drive/folders/1Gy9MKHwWgql2Bpu28xJtZg6IAgV2TTyR

spatcho, 2025-04-04:
I can happily say that I see now grbl response messages - error messages but at least it communicates

spatcho, 2025-04-04:
> On thermal printer we always recommend to use the TX pin and GND only because Krux doesn't need to receive nothing from the printer, and connecting RX sometimes could cause issues.

For the moment, I am connecting 5v gnd rx tx to the cnc controller, without external power, usbc on the wondermv, that way the wondermv is powering the controller and it easy the initial testing. I did try few times with external power to the cnc controller, that way it powers the wondermv also, it went ok. Both devices powered seems fine too. The cnc controller is a blackbox x4 from openbuilds. Both rx and tx are needed.

tadeubas, 2025-04-04:
I don't know too much about the board, but I think it was not designed to be powered by the UART interface connector, I would plug in the USB cable anyway. Also I would not power the controller using the UART power only, I would always connect the external power supply. Nice to see your progress, thx for sharing! 😁

spatcho, 2025-04-05:
Pull request done. I did adapt and test all : to file, to serial, to svg. As noted on cnc.py : This has been tested on the machine without router started and without bit, only to see that it move correctly and that it complete a full cycle. 

About powering the wondermv from uart without usb : sometimes it doesn't start and need usb, but I could complete a full cycle without usb. Maybe powering the router will introduce issues but not sure as this serial connection is used to power the a remote controller based on esp32 while the cnc is milling (it's called the blackbox interface controller).

About powering the grbl controller from the uart : that was not the intention at the beginning but I noticed it worked and it helped me for initial testing, that way I didn't have to be close the cnc machine. But sure it's not a real life scenario.

spatcho, 2025-04-05:
First real milling but it's not recognized by krux scanning : is it because it's white dots on black background ? Or you see something else ? It's a compact qr of "action action ..."

[image]

spatcho, 2025-04-05:
I tried milling on aluminium, and painting the dots in dark, also couldn't scan it. I guess it really want squares dots, but I don't have bits smaller than 3.175 mm.and I don't really want to make the qr code bigger. Ultimately what I want is to keep my keys safe on a stainless steel media. I may replace the cnc router by a laser module and try engraving. I don't have laser module yet.

tadeubas, 2025-04-05:
I think it needs a white background in order to be detected, and maybe the dots need to be darker

tadeubas, 2025-04-05:
Here we have examples of metal plates, but we use a custom code to detect those as they are not QR codes...: https://selfcustody.github.io/krux/getting-started/features/tinyseed/

odudex, 2025-04-07:
> is it because it's white dots on black background ? It need a border ? Or you see something else ?

Really cool!
But krux does not read inverted colors, it needs light background and black features. QR codes do not need a border, just a "quiet zone", >= 2 blocks, which your QR has.
Dots can be used in data area, but the if the alignment stones at the corners had straight lines it would detect QRs much easier. Do you think you can program dedicated moves for them? A smaller bit could also help it be less bumpy.

jdlcdl, 2025-04-07:
Could you spray paint the whole thing with white, let it dry, then spread black paint over it and wipe it off the high surface before it dries?

Same on alu or inox, but could do a light sanding on high surface to get it "light" again?

spatcho, 2025-04-08:
> Could you spray paint the whole thing with white, let it dry, then spread black paint over it and wipe it off the high surface before it dries?
> 
> Same on alu or inox, but could do a light sanding on high surface to get it "light" again?

Thanks for the tip, wiping before it dry I didn't think about, will try (need to buy the white paint)

spatcho, 2025-04-08:
> Really cool! But krux does not read inverted colors, it needs light background and black features. QR codes do not need a border, just a "quiet zone", >= 2 blocks, which your QR has. Dots can be [...]

Thanks for the info. I used black because that is what I had, my bad. I will get the tools (white paint, smaller bit, etc) and try again.

spatcho, 2025-04-08:
I see the old code was supported the invert mode, I didn't keep it, I will put it back and see how it goes.

spatcho, 2025-04-08:
> I see the old code was supporting the invert mode, I didn't keep it, I will put it back and see how it goes.

Success this time using inverted mode. When inverted, additional to the original code, I add a supplementary margin of one cell width all around the original qr_code, otherwise it cannot be scanned. Result is 96x96 mm wide, milled using a 2mm bit. 

[image]

Will publish the changes after I do some more work on the grbl printer (direct to serial)

spatcho, 2025-04-08:
> > I see the old code was supporting the invert mode, I didn't keep it, I will put it back and see how it goes.
> 
> Success this time using inverted mode. When inverted, additional to the original code, I add a supplementary margin of one cell width all around the original qr_code, otherwise it cannot be scanned. Result here is 96x96 mm, milled using a 2mm bit.
> 
> [image]
> 
> Will publish the changes after I do some more work on the grbl printer (direct to serial)

All published

spatcho, 2025-04-09:
A 50x50 with 2mm bit, inverted, was recognized also 

[image]

spatcho, 2025-04-09:
So it work pretty well. I think best would be laser engraving. Don't have one at hand but it's on my purchase wish list.

odudex, 2025-04-09:
Beautiful! Excellent work!

spatcho, 2025-04-14:
Laser fun

[image]

tadeubas, 2025-04-14:
Wow, very nice! I've completed the docs about the CNC engraving with drill bit part, it is on #565 

Laser needed another code and driver? Can you commit?

The "current" docs:

[image]

spatcho, 2025-04-15:
Stainless steel laser engraving

Code was modified to add laser specific parameters and limit movements to xy plane. It has been committed to qr_cnc_svn branch on spatcho fork.

[image]

From left to right : 
- 34x34 mm, 50% power, could be scanned after playing with light conditions
- 21x21 mm, 5% power, couldn't be scanned
- 21x21 mm, 50% power, couldn't be scanned

spatcho, 2025-04-15:
I am ok to delete my existing fork and pr's, recreate a fork based on develop branch, and create one pr for svg, one pr for cnc and one pr for laser, or maybe one pr for svg and one pr for cnc + laser, as it is very similar, both are cnc, one is with a router, one is with a laser.

qlrd, 2025-04-15:
> I am ok to delete my existing fork and pr's, recreate a fork based on develop branch, and create one pr for svg, one pr for cnc and one pr for laser, or maybe one pr for svg and one pr for cnc + laser, as it is very similar, both are cnc, one is with a router, one is with a laser.

@spatcho I already created one for CNC (see #562) with minor adjustments about format, linting and tests.

odudex, 2025-04-15:
Awesome! I'm curious about how the engraving behaves regarding attacks with scratches, chemicals, heat, etc. From things I viewed with similar engraving it's pretty robust! Of course not as good as a machined relief, but much better than paper or plastic. You should do some videos about it!

odudex, 2025-04-15:
Regarding not being able to scan smaller ones, focus can play a role. To scan smaller objects you need to get the device closer, which requires a different focus adjust. Zoomed mode may be helpful.

spatcho, 2025-04-15:
> Awesome! I'm curious about how the engraving behaves regarding attacks with scratches, chemicals, heat, etc. From things I handled with similar engraving it's pretty robust! Of course not as good as a machined relief, but much better than paper or plastic. You should do some videos about it!

Depends on laser power. With mine, on stainless steel, it's supposed to be deeper and better than just etching. I think we are scratch proof, I tried with knife and metal brush. Heat we should be good, I may try against flame. Chemicals I don't have chemicals suitable to do that and don't really want to buy and store such products. Will try water because laser burn and burned stainless steel lose rust protection, at least in welding. I guess no interest to test on non metallic medias. The blue laser I have cannot engrave bare metals other than steel (stainless or not).

> > I am ok to delete my existing fork and pr's, recreate a fork based on develop branch, and create one pr for svg, one pr for cnc and one pr for laser, or maybe one pr for svg and one pr for cnc + laser, as it is very similar, both are cnc, one is with a router, one is with a laser.
> 
> [@spatcho](https://github.com/spatcho) The CNC was already merged (see [#562](https://github.com/selfcustody/krux/pull/562)) with minor adjustments about format, linting, tests and coverage. If you want, i can do the same for SVG and laser.
> 
> I will ask to not delete yet, if you agree.
> 
> I think more 2 PRs are better (svg, laser) for "reviewability"

At the time of writing when I checked it wasn't merged onto develop, now I see it is. Will test it. I prefer to make the PRs. When to delete ?

spatcho, 2025-04-15:
> [@spatcho](https://github.com/spatcho) The CNC was already merged (see [#562](https://github.com/selfcustody/krux/pull/562)) with minor adjustments about format, linting, tests and coverage. If you [...]

At the time of writing when I checked it wasn't merged onto develop, now I see it is. Will test it. I prefer to make the PRs. When to delete ?

qlrd, 2025-04-15:
> At the time of writing when I checked it wasn't merged onto develop, now I see it is. Will test it. I prefer to make the PRs. When to delete ?

I think its better to delete when fully merged (one for svg and one for laser).

spatcho, 2025-04-15:
> Regarding not being able to scan smaller ones, focus can play a role. To scan smaller objects you need to get the device closer, which requires a different focus adjust. Zoomed mode may be helpful. Scanning the screen, I get big and small at the first frame. Except the middle faded one.

Thanks about zoomed mode, still no luck, even on screen, but will try again in day light

tadeubas, 2025-04-15:
> > Awesome! I'm curious about how the engraving behaves regarding attacks with scratches, chemicals, heat, etc. From things I handled with similar engraving it's pretty robust! Of course not as good as a machined relief, but much better than paper or plastic. You should do some videos about it!
> 
> Depends on laser power. With mine, on stainless steel, it's supposed to be deeper and better than just etching. I think we are scratch proof, I tried with knife and metal brush. Heat we should be good, I may try against flame. Chemicals I don't have chemicals suitable to do that and don't really want to buy and store such products. Will try water because laser burn and burned stainless steel lose rust protection, at least in welding. I guess no interest to test on non metallic medias. The blue laser I have cannot engrave bare metals other than steel (stainless or not). Will see for the vids yes.

You can use just vinegar or pure lemon juice. Leave the part with some samples (100%, 75%, 50%, 25% laser strength) submerged in a sealed plastic container for months, it should be enough to replicate stronger acids.

spatcho, 2025-04-15:
> You can use just vinegar or pure lemon juice. Leave the part with some samples (100%, 75%, 50%, 25% laser strength) submerged in a sealed plastic container for months, it should be enough to [...]

Ok will do that

spatcho, 2025-04-15:
> Regarding not being able to scan smaller ones, focus can play a role. To scan smaller objects you need to get the device closer, which requires a different focus adjust. Zoomed mode may be helpful. Scanning the screen, I get big and small at the first frame. Except the middle faded one.

What device did you run krux on ?

kkdao, 2025-04-15:
> What device did you run krux on ?

Odudex has all the devices:) That being said, my guess is it was either the Amigo, WonderMV, or Yahboom used to scan the screen. Hopefully not the Amigo as you won't be able to test with that, it's almost impossible to procure these days!

[7 more replies]

PR #562 (2025-04-15): CNC print export
### What is this PR for?

Fix #550

### Changes made to:
- [x] Code
- [x] Tests
- [ ] Docs
- [x] CHANGELOG

### What is the purpose of this pull request?
- [x] Bug fix
- [ ] New feature
- [ ] Docs update
- [ ] Other

### Description

Print qr to cnc, write to file:

* adapt code to latest qr_code array structure;
* print qr to cnc, write directly to cnc;
* uncomment, test, and adapt code to fit an openbuilds grbl 1.1 cnc (see comments on cnc.py);
* added information on CHANGELOG about the changes.

### Context

@spatcho created a previous PR in #557 (and in truth, he's the aut [... cut]
commit b58a8a80 (2025-04-15): CNC print export (#562)

Fix #550

Print qr to cnc, write to file:

* adapt code to latest qr_code array structure;
* print qr to cnc, write directly to cnc;
* uncomment, test, and adapt code to fit an openbuilds grbl 1.1 cnc;
* added information on CHANGELOG about the changes.
closed 2025-09-15 by odudex