selfcustody/krux #568: [Other] alt_wordlist parameter deadcode on src/pages/mnemonic_editor.py

qlrd, 2025-04-16:
#### Summary

The `alt_wordlist` parameter in `MnemonicEditor` methods such as `compute_search_ranges()`, `autocomplete()`, and `autocomplete()` is never used effectively during interactive flows (`edit()` and `edit_word()`).

#### Details

Although several methods accept `alt_wordlist`, the actual mnemonic editing flow never passes it down properly. For example:

```diff
def edit_word(self, index):
    """Edit a word"""
    word_txt = str(index + 1) + ". " + self.current_mnemonic[index]
    self.flash_text(word_txt)
    while True:
+        self.compute_search_ranges()
        # if new and last word, lead input to a valid mnemonic
        if self.new_mnemonic and index == self.mnemonic_length - 1:
            final_words = Key.get_final_word_candidates(self.current_mnemonic[:-1])
+            word = self.capture_from_keypad(
+                t("Word %d") % (index + 1),
+                [LETTERS],
+                lambda x: self.autocomplete(x, final_words),
+                lambda x: self.possible_letters(x, final_words),
+            )
        else:
+            word = self.capture_from_keypad(
+                t("Word %d") % (index + 1),
+                [LETTERS],
+                self.autocomplete,
+                self.possible_letters,
+            )
        if word == ESC_KEY:
            return None
        if word in WORDLIST:
            return word
        word = ""
```

Here, `self.compute_search_ranges`, `self.autocomplete` and `possible_letters` are called without any `alt_wordlist`, meaning the fallback BIP39 WORDLIST is always used internally. This renders the ****`alt_wordlist` parameter dead code****.

#### Why It Matters

1. Dead Logic: The `alt_wordlist path`, including important logic like:

```diff
def autocomplete(self, prefix, alt_wordlist=None):
        """Autocomplete a word"""
        if alt_wordlist:
            wordlist = alt_wordlist
            self.compute_search_ranges(wordlist)
        else:
            wordlist = WORDLIST
        if len(prefix) > 0:
            letter = prefix[0]
+            if letter not in self.search_ranges:
+                return None
            start, stop = self.search_ranges[letter]
            matching_words = list(
                filter(
                    lambda word: word.startswith(prefix),
                    wordlist[start:stop],
                )
            )
            if len(matching_words) == 1:
                return matching_words[0]
        return None
```

is never triggered unless manually forced in tests (see example code at [reproduct](#reproduct) below.

2. False Coverage: Tests may assume alternate wordlists are in use, but actual calls always fallback to english BIP39.
 
3. Project Scope: If the project does not plan to support non-BIP39 wordlists, this param is unnecessary.

So i think that just remove this parameter is better than create unecessary tests.

#### Reproduct

This test makes a lot of patches in `MnemonicEditor` to cover the already mentioned `autocomplete` method:

```diff
diff --git a/tests/pages/test_mnemonic_editor.py b/tests/pages/test_mnemonic_editor.py
index b9d0db2c..048b30e0 100644
--- a/tests/pages/test_mnemonic_editor.py
+++ b/tests/pages/test_mnemonic_editor.py
@@ -5,6 +5,22 @@ TEST_12_WORD_MNEMONIC = (
 )
 TEST_24_WORD_MNEMONIC = "brush badge sing still venue panther kitchen please help panel bundle excess sign couch stove increase human once effort candy goat top tiny major"
 MNEMONICS = [TEST_12_WORD_MNEMONIC, TEST_24_WORD_MNEMONIC]
+CUSTOM_WORDLIST = [
+    "academic",
+    "banjo",
+    "blooming",
+    "cobweb",
+    "glimmer",
+    "harvest",
+    "heaven",
+    "jackal",
+    "myriad",
+    "nectar",
+    "quantum",
+    "salary",
+    "sharply",
+    "stinger",
+]

 def test_confirm_given_words(mocker, multiple_devices):
@@ -266,3 +282,32 @@ def test_button_used_but_touch_enabled(mocker, amigo):
     mnemonic_editor._map_words()
     ctx.input.touch.x_regions.append.assert_called()
     ctx.input.touch.y_regions.append.assert_called()
+
+
+def test_edit_mnemonic_autocomplete_prefix_not_in_search_ranges(mocker, amigo):
+    from krux.pages import ESC_KEY
+    from krux.pages.mnemonic_editor import MnemonicEditor
+
+    ctx = create_ctx(mocker, [])
+    mnemonic_editor = MnemonicEditor(ctx, TEST_12_WORD_MNEMONIC)
+
+    # patch autocomplete method to enforce using the CUSTOM_WORDLIST
+    def patched_autocomplete(prefix, alt_wordlist=None):
+        return MnemonicEditor.autocomplete(mnemonic_editor, prefix, CUSTOM_WORDLIST)
+
+    # patch possible_letters as well just for coverage
+    def patched_possible(prefix, alt_wordlist=None):
+        return MnemonicEditor.possible_letters(mnemonic_editor, prefix, CUSTOM_WORDLIST)
+
+    # patch capture_from_keypad to simulate 'z' input and capture autocomplete result
+    def patched_keypad(label, input_type, autocomplete_fn, possible_letters_fn):
+        result = autocomplete_fn("z")  # triggers actual coverage path
+        assert result is None  # confirms the path was taken
+        return ESC_KEY
+
+    mnemonic_editor.autocomplete = patched_autocomplete
+    mnemonic_editor.possible_letters = patched_possible
+    mnemonic_editor.capture_from_keypad = patched_keypad
+
+    result = mnemonic_editor.edit_word(0)
+    assert result is None
```

jdlcdl, 2025-04-16:
It's common to bulletproof code while writing, but I agree that it's better to delete code that isn't needed... as soon as we're sure.

How are you finding all of these?  Are they just missing test coverage and then you realize that you have to jump thru hoops in order to cover them?  Anyways, I like this work as long as the code is never reached.

qlrd, 2025-04-17:
> How are you finding all of these?

@jdlcdl, i found it when trying to cover the lines 90 and 113.  

> Are they just missing test coverage and then you realize that you have to jump thru hoops in order to cover them?

When making a lot of patches in the `MnemonicEditor` class until i patched `autocomplete` and `possible_letters` before use `edit`, as well use words that were outside the WORDLIST. 

("If I'm doing that, there's probably something wrong with this approach.") 

Then i read file where `WORDLIST` is defined and realized that, even when user chooses the last word, always will exist a word to be choosed, unless another WORDLIST is provided without those on BIP39. Therefore, theoretically, such lines would never be covered. At least with the current approach in `MnenomicEditor`.

> It's common to bulletproof code while writing

I thought that too, as a defensive means, correct? Maybe there is some way to maintain this defensive approach, otherwise maintaining coverage in a real situation and not forced through patches?

qlrd, 2025-04-17:
Tested some times with simulator and nothing went wrong. I will try with real device, but it would be nice if you guys can reproduce.

jdlcdl, 2025-04-18:
I think you're doing the right thing here.  It's natural to try to bullet-proof while "creatively" writing but you're find where that those lines are of a type: "you're not going to need it" and so they don't need to be there.  Lines of code deleted are lines that don't need to be maintained.

tadeubas, 2025-04-30:
PR #569 is merged into develop. Can we close the issue?

PR #569 (2025-04-25): Remove deadcode in src/pages/menomic_editor.py
### What is this PR for?

Fix #568

### Changes made to:
- [x] Code
- [ ] Tests
- [ ] Docs
- [ ] CHANGELOG

### What is the purpose of this pull request?
- [ ] Bug fix
- [ ] New feature
- [ ] Docs update
- [x] Other: remotion of possible deadcode.

### Details

The `alt_wordlist` parameter in `MnemonicEditor` methods such as `compute_search_ranges()`, `autocomplete()` , and `possible_letters()`  is only meaningfully used in one specific case: when editing the final word of a new mnemonic.

In all other interactive flows (`edit()` , `edit_word()`), [... cut]
commit dabca168 (2025-04-25): Remove deadcode in src/pages/menomic_editor.py (#569)

Fix #568

The alt_wordlist parameter in MnemonicEditor methods such as
compute_search_ranges(), autocomplete(), and possible_letters() is only
meaningfully used in one specific case: when editing the final word of a
new mnemonic.

In all other interactive flows (edit(), edit_word()), it is ignored,
making calls such `if letter not in self.search_ranges` effectively dead
— unless forcefully tested with patched inputs.

The virtual keyboard only presents valid next letters (from
possible_letters()), which ensures users cannot enter a letter  [... cut]