selfcustody/krux #572: [Enhancement] backup seed with descryptor together.

Garrysoon, 2025-04-18:
that would be great to  along with wallet descriptor to eliminate manual setup for every sign in multisign.

comassky, 2025-04-18:
To add a little more detail, I think it would be useful to store in the QR Code (and load / use when scanning) :

- The network
- The Policy Type (Single Sig / MultiSig / Miniscript and subconfiguration ( 44 vs 49 vs 84 vs 86 vs 49) associated
- Wallet descriptor if multisig

qlrd, 2025-04-18:
> that would be great to reserve mnemonic phrase along with wallet descryptor via QR/file to eliminate manual setup for every sign in multisign.

Currently, IMO, this could be hard at the current stage with QRCodes. Especially for those who use printing from the device. With files is somewhat easier (perhaps with a feature in development, exporting a QRCode to SVG could be plausible).

> To add a little more detail, I think it would be useful to store in the QR Code (and load / use when scanning)
>
> * The network
> * The Policy Type (Single Sig / MultiSig / Miniscript and subconfiguration ( 44 vs 49 vs 84 vs 86 vs 49) associated
> * Wallet descriptor if multisig

These infos are already described in descriptors.

> However, I think the addition of the Wallet descriptor will cause problems with QR Code sizes, making scanning perhaps too difficult for the resolution of the internal camera.

A descriptor in a QRcode requires a lot of space (once tried a 2-of-3 multisig and simple inheritance Liana's scheme -- imagine expanding multisig -- backup with printed QRCode and was impossible to scan). 

It might be interesting to wait for a [standardization of this](https://delvingbitcoin.org/t/a-simple-backup-scheme-for-wallet-accounts/1607).

Garrysoon, 2025-04-18:
Then maybe we should focus on the saved encrypted mnemonic file and add a default descriptor there?