selfcustody/krux #578: [Enhancement] summarize the new feature or enhancement

qlrd, 2025-04-19:
#### Describe your request
 
#577 updated the PR template by adding an important verification step to PR reviews: asking whether the developer built the firmware.

While this is very important, it lacks a "don't trust, verify" approach. That is, someone can say "yes, I built it". But how can we be sure? There is a way: provide the `sha256sum` hash of the build, so that it can be reproduced by reviewers.

This issue proposes some additional checks (worth changing):

```markdown
### Did you build the code and tested on device?
- [ ] Yes
- [ ] No

### In which device you built?
- [ ] M5stickV
- [ ] Amigo
- [ ] Bit
- [ ] Docker
- [ ] Cube
- [ ] Yahboom
- [ ] WonderMV

Please provide the `sha256sum` hash of the build: <!-- Put here the hash -->
```

This may make the process more bureaucratic, but I believe it will also provide more certainty about what is being reviewed.

tadeubas, 2025-04-19:
Code hashes are useless in development because temporary changes and debug prints alter them. Requiring hashes per commit just creates busywork without real value. What matters is ensuring changes are properly tested, not matching hashes.

Also, the option "no" in the question it is redundant, if you did not checked "yes" it already implies a "no".

tadeubas, 2025-04-19:
List all devices to be checked could be useful only before a release, not development. For the template we could just change the "Yes" to "Yes, build and tested on "

tadeubas, 2025-04-30:
PR #580 is merged into develop. Can we close the issue?

PR #580 (2025-04-25): Followup of PR 557
### What is this PR for?

Fix #578

### Changes made to:
- [ ] Code
- [ ] Tests
- [ ] Docs
- [ ] CHANGELOG
- [x] Other: PR/issue templates

### Did you build the code and tested on device?
- [ ] Yes
- [x] No

### In which device you built?
- [ ] M5stickV
- [ ] Amigo
- [ ] Bit
- [ ] Docker
- [ ] Cube
- [ ] Yahboom
- [ ] WonderMV

### What is the purpose of this pull request?
- [ ] Bug fix
- [ ] New feature
- [ ] Docs update
- [x] Other: PR template

### Additional context:

While this is very important to ask if the developer built the firmware on a device, it lacks more info [... cut]
closed 2025-05-06 by odudex