selfcustody/krux #708: [Enhancement] Secure boot

Sami32, 2025-09-13:
#### Describe your request
It will be great if secure boot could be added like with the Jade DIY firmware.

Sami32, 2025-10-26:
https://rp.os3.nl/2019-2020/p49/presentation.pdf

Oh...i see, i'll try to get more specific documentations on the Kendryte K210 because it seem not widely accessible as for now?

odudex, 2025-10-26:
K230 has it working, I got some devices with it and tested them. But the English documentation is poor, and the python environments available are too big (more than 200MB) and closed source. K230 targets Linux OS use...
I'm investing time on ESP32P4, which has secure boot (and other interesting security fetures) and differently from other ESP32s, does not have wi-fi and BLE.
https://github.com/odudex/crux

odudex, 2025-10-30:
> In case a K210 device have OTP not disabled with a factory-set flag before shipping, then here is the code:
> 
> [otp.c](https://github.com/user-attachments/files/23237900/otp.c) [otp.h.zip](https://github.com/user-attachments/files/23237960/otp.h.zip)

Where did this come from? Any context?