selfcustody/krux #724: [Bug] Mismatch with the Specification (KEF Encryption Format)

earthdiver, 2025-09-20:
In the compression mode of the KEF Encryption Format, the window size used by the Deflate algorithm differs between the specification (`zlib.compress` with `wbits=-10`) and the Krux implementation (default value in `DeflateIO`, equivalent to `-8`).
Because of this, there is a risk that data compressed with implementations other than Krux may not be decompressible in Krux.

**Proposed Fix 1:**

Explicitly specify `wbits=-10` for both compression and decompression.
 **Pros:** Matches the specification.
 **Cons:** Maintains compatibility in the sense that data compressed with the current version can still be decompressed.
  However, full compatibility is not achieved since the compression results differ between the current and revised versions.

**Proposed Fix 2:**

Explicitly specify `wbits=-10` only for decompression.
 **Pros:** Ensures full compatibility with the current version.
 **Cons:** Leaves a discrepancy with the specification regarding compression.

**Device(s) affected**
- [x] Amigo
- [x] M5stickV
- [x] Cube
- [x] Dock
- [x] Bit
- [x] Yahboom
- [x] WonderMV

**Version affected**
- [x] Official release (selfcustody/krux): vXX.YY.Z 
- [x] Beta release (odudex/krux_binaries): vXX.YY.Z-betaWW

tadeubas, 2025-09-20:
I think there’s a misunderstanding here:

The decompressor (inflate) does not depend on the window size chosen by the compressor (deflate), as long as the compressed stream is valid. Deflate encodes the effective window size into the stream itself, so any compliant decompressor can handle it.

The only consequence of using `-8` vs. `-10` is a difference in compression efficiency, not in decompression compatibility. Data compressed with either will always decompress correctly.

So the claim that "data compressed with implementations other than Krux may not be decompressible in Krux" isn’t accurate. The real point is that if the spec says `-10` but Krux uses `-8`, then Krux isn’t spec-compliant, and compressed outputs won’t match bit-for-bit. But that should be fine, since KEF doesn’t require bit-for-bit compressed envelopes.

jdlcdl, 2025-09-20:
While adding compressed versions, I'd borrowed compression params from the krux BBQr module, which intended to be compatible with coldcard's BBQr specifications.   We learned that MaixPy's deflateIO and python's zlib(wbits=-10) were NOT binary compatible during early testing of KEF between our simulator (zlib) and devices.

While I never dove-in to try to resolve binary compatibility, I did extensive testing to be sure that we can `kef.reinflate()` from different implementations, and I left a trace of doing so within the unit-test module, see: `tests/test_kef.py::test_brute_force_compression_checks`.

During testing, I fiddled with the size of input-text to be compressed and the number of samples.  We can also adjust whether this test creates its own input, or reads a file on sdcard, or writes its own input to a file on sdcard) so that we can move the `/sd/brute_force.txt` to a different device for testing `kef.reinflate()` from some other implementation.

Krux's on-device tests compare a single-byestring each, of compressed-bytes sample, created with different implementations (one for MaixPy, one for zlib, more in the future as we find implementations that are not binary compatible).

Thank you everyone for taking a look around this issue.

earthdiver, 2025-09-20:
> I think there’s a misunderstanding here:
>
> The decompressor (inflate) does not depend on the window size chosen by the compressor (deflate), as long as the compressed stream is valid. Deflate encodes [...]

Is that really the case even in RAW format? In what situations does the "invalid distance too far back" error occur? Also, what is the purpose of the wbits argument during decompression?

By the way, according to what I checked with ChatGPT, reproducing the "invalid distance too far back" error requires at least a 16KB payload (though I didn’t verify this myself by writing a program).

earthdiver, 2025-09-20:
Since 16KB is the size only when specially crafted data is used to reproduce the error, it does seem certain that this occurs only rarely.

jdlcdl, 2025-09-20:
Can you use kef._reinflate() with this text?  I'm getting an error "invalid stored blocks length" even with your implementation.

I did like
```python
import kef  # earthdiver1's implementation

with open("ascii_plain_min.txt", "r") as h:
      contents = h.read()

kef._reinflate(contents.encode())  # error
kef._reinflate(contents.encode("latin-1"))  # still error
```

Once I can get it working in normal python, so that I know what to expect, I'll try it in a miniterm terminal connection to a k210 MaixPy device.

earthdiver, 2025-09-20:
> From what I understood, the sample is not compressed. Does the error happen when decompressing after it's compressed with wbits=-15?

Yes

earthdiver, 2025-09-20:
Since ChatGPT cannot run MicroPython, it tested decompression with zlib. As DeflateIO’s Deflate seems to be a custom implementation, it might not be possible to reproduce the issue accurately.

jdlcdl, 2025-09-20:
I don't know how to do it, but I'll ask about leaning towards trying to use a 10bits window in DeflateIO, since that matches current KEF specifications, and also BBQr (suspecting its compression may gain more industry traction with time on microcontrollers)

earthdiver, 2025-09-20:
> Using ascii_plain_min2.txt as binary contents (read it as text, encode(utf8))

Please read it as binary (in "rb" mode) and do not use .encode().

earthdiver, 2025-09-20:
The file size is large, but if you have capacity, please also try this data. (Compress with wbits=-10 and decompress with wbits=-8.)

[ascii_plain_min3.txt](https://github.com/user-attachments/files/22443535/ascii_plain_min3.txt)

jdlcdl, 2025-09-20:
> The file size is large, but if you have capacity, please also try this data. (Compress with wbits=-10 and decompress with wbits=-8.)
> 
> [ascii_plain_min3.txt](https://github.com/user-attachments/files/22443535/ascii_plain_min3.txt)

I've tried the third ascii_plain_min3.txt file.

While large, it compressed lots in normal python and less efficiently in krux devices (705 bytes instead of 248 with zlib), but all kef implementations were able to reinflate and recover the original 31776 bytes.  I tried compressing in krux, then decompressing with wbits= -10, -15, and vice versa.

earthdiver, 2025-09-21:
I also tried verifying the reproducibility of the error when compressing and decompressing with zlib on both ends. I found that even with the same zlib, the behavior differs depending on how it is implemented.

[verify_zlib_raw_wbits.py](https://github.com/user-attachments/files/22448818/verify_zlib_raw_wbits.py)

**Execution Result**
```
Compression: completed (wbits=-15, raw DEFLATE)
Input (uncompressed): 16385 bytes
Output (compressed):   13579 bytes
Baseline: successfully decompressed using wbits=-10
Case 1: unexpectedly succeeded with wbits=-8 (no error)
Case 2: unexpectedly succeeded with wbits=-8 (no error)
Case 3: expected failure with wbits=-8 -> Error -3 while decompressing data: invalid distance too far back
Case 4: expected failure with wbits=-8 -> Error -3 while decompressing data: invalid distance too far back
```

earthdiver, 2025-09-21:
With this data(33KB), decompression fails in all cases (1–4) with the error 'invalid distance too far back'.

[ascii_plain_min4.txt](https://github.com/user-attachments/files/22449190/ascii_plain_min4.txt)

tadeubas, 2025-09-21:
> So, wouldn't it be the case to just warn on the spec to **not use wbits higher than 10**

Yes, I believe we have to enforce this limit.

jdlcdl, 2025-09-22:
@earthdiver1, Thank you for raising this issue and being so helpful in clarifying/testing KEF.

Note that Odudex has already merged improvements to krux's micropython implementation to bulletproof the default of 10bits window for decompressing in DeflateIO.

Also note that you've been very helpful in a PR I just issued, which improves a test in the KEF test-suite, and help illustrate this important compression param... and also includes samples so that we can see it in action.

I'll note that I see your pr to 3rdIteration's seedsigner fork, so one more "Thank you" for [this](https://github.com/3rdIteration/seedsigner/pull/186)